wollewald/ADS1115_WE #7: SPS is Low

bgondell, 2020-08-07:
Hi!!

I'm testing this library, but I can't got the 860SPS, using SingleShot and by reading 0_GND, 1_GND and 2_GND, 3_GND and repeating. SPS is 285 - 307.
Even if I only take only one of them in SingleShot, SPS rises only up to 333.
Also I'm getting some crosstalk I think, and using a capacitor isn't an option since I'm measuring AC voltaje.

Any ideas I can try??

Thanks a lot!!

```cpp
#include<ADS1115_WE.h> 
#include<Wire.h>
ADS1115_WE ads1(0x48);

void setup() {
  // put your setup code here, to run once:
  Wire.begin();
  
  Serial.begin(115200);

  if(!ads1.init()){
    Serial.println("ADS1 not connected!");
  }
  ads1.setVoltageRange_mV(ADS1115_RANGE_4096);
  ads1.setConvRate(ADS1115_860_SPS);

  delay(1000);
}

void loop() {

  long start = millis();
  
  double sample1 = readADC(ads1, ADS1115_COMP_0_GND);

  double sample2 = readADC(ads1, ADS1115_COMP_1_GND);

  double sample3 = readADC(ads1, ADS1115_COMP_2_GND);

  double sample4 = readADC(ads1, ADS1115_COMP_2_GND);

  Serial.println("Analog 0: "); Serial.println(sample1);
  Serial.print("Analog 1: "); Serial.println(sample2);
  Serial.print("Analog 2: "); Serial.println(sample3);
  Serial.print("Analog 3: "); Serial.println(sample4);
  Serial.print("SPS :"); Serial.println(1000*4/(millis()-start));
  

}

float readADC(ADS1115_WE adc, ADS1115_MUX channel) {
  float voltage = 0.000;
  adc.setCompareChannels(channel);
  adc.startSingleMeasurement();
  while(adc.isBusy()){}
  voltage = adc.getResult_mV(); // alternative: getResult_mV for Millivolt
  return voltage;
}

```

wollewald, 2020-08-07:
Interesting one! The answer is you are measuring mainly the speed of the Serial.print / Serial.println function and the I"C connection.

1) Serial.print / Serial.println
I took out everything related with the ADS1115 after the setup. The readADC only returns 2.0 (just any number). The "SPS" for this was 666 on an Arduino Uno, which is amazingly slow. 

```cpp
#include<ADS1115_WE.h> 
#include<Wire.h>
ADS1115_WE ads1(0x48);

void setup() {
  // put your setup code here, to run once:
  Wire.begin();
  
  Serial.begin(115200);

  if(!ads1.init()){
    Serial.println("ADS1 not connected!");
  }
  ads1.setVoltageRange_mV(ADS1115_RANGE_4096);
  ads1.setConvRate(ADS1115_860_SPS);

  delay(1000);
}

void loop() {

  long start = millis();
  
  double sample1 = readADC(ads1, ADS1115_COMP_0_GND);

  double sample2 = readADC(ads1, ADS1115_COMP_1_GND);

  double sample3 = readADC(ads1, ADS1115_COMP_2_GND);

  double sample4 = readADC(ads1, ADS1115_COMP_2_GND);

  Serial.println("Analog 0: "); Serial.println(sample1);
  Serial.print("Analog 1: "); Serial.println(sample2);
  Serial.print("Analog 2: "); Serial.println(sample3);
  Serial.print("Analog 3: "); Serial.println(sample4);
  Serial.print("SPS :"); Serial.println(1000*4/(millis()-start));
  

}

float readADC(ADS1115_WE adc, ADS1115_MUX channel) {
//  float voltage = 0.000;
//  adc.setCompareChannels(channel);
//  adc.startSingleMeasurement();
//  while(adc.isBusy()){}
//  voltage = adc.getResult_mV(); // alternative: getResult_mV for Millivolt
  return 2.0;
}
```

2) without Serial.print/println
In a second experiment I took out all Serial.print / println (between the measures).  The sketch does 4 times 860 mesasurements and determines the time needed. Should be 1 second (after divided by 4), but it needed 2.69. 

```cpp
#include<ADS1115_WE.h> 
#include<Wire.h>
ADS1115_WE ads1(0x48);

void setup() {
  // put your setup code here, to run once:
  Wire.begin();
  
  Serial.begin(115200);

  if(!ads1.init()){
    Serial.println("ADS1 not connected!");
  }
  ads1.setVoltageRange_mV(ADS1115_RANGE_4096);
  ads1.setConvRate(ADS1115_860_SPS);

  delay(1000);
}

void loop() {

  long start = millis();
  
  for(int i=0; i<860; i++){
    double sample1 = readADC(ads1, ADS1115_COMP_0_GND);
  
    double sample2 = readADC(ads1, ADS1115_COMP_1_GND); 

    double sample3 = readADC(ads1, ADS1115_COMP_2_GND);
  
    double sample4 = readADC(ads1, ADS1115_COMP_2_GND);
}
  
  long timefor860Samples = (millis()-start)/4;
  Serial.print("Time needed 860 samples:"); Serial.println(timefor860Samples);
}

float readADC(ADS1115_WE adc, ADS1115_MUX channel) {
  float voltage = 0.000;
  adc.setCompareChannels(channel);
  adc.startSingleMeasurement();
  while(adc.isBusy()){}
  voltage = adc.getResult_mV(); // alternative: getResult_mV for Millivolt
return voltage;
}
```
3) I2C connection
In a third experiment I have taken the last sketch and just took out the measurement itself. The result was 1.35 seconds. The difference is 1.34 seconds which is the measurement time plus the time to send the instruction to measure. 
```cpp
float readADC(ADS1115_WE adc, ADS1115_MUX channel) {
  float voltage = 0.000;
  adc.setCompareChannels(channel);
//  adc.startSingleMeasurement();
//  while(adc.isBusy()){}
  voltage = adc.getResult_mV(); // alternative: getResult_mV for Millivolt
return voltage;
}
```

If you want to speed up the time do use the alert pin for conversion ready check. Have a look into the example sketch "Single_Shot_Conv_Ready_Alert_Controlled.ino". Long answer, but I was also keen to look into the details where all the time is gone.

bgondell, 2020-08-10:
Thanks for everything! I will continue to do more testing ... And Damn! I didn't route the alert pin on the board as I thought I wouldn´t need it.

wollewald, 2020-08-10:
Good luck!

closed 2020-08-10 by wollewald