wollewald/ADS1115_WE #12: Can you make an example on how to use the library in continous mode using more than one input?

Potul72, 2020-09-23:


wollewald, 2020-09-23:
@Potul72; I am hesitating to do that because I do not recommend to use the continuous mode when you want to change channels. There is only one register for the conversion results and if you request it with `getResult_V()` or `getResult_mV()` you will get what is currently in the register even if a conversion is ongoing. So it could happen that you change the channel request a result and still get the result from the former channel. The `isBusy()` function which is used to wait for the latest conversion in the single shot mode does not work in the continuous mode. 

If you really want to use the continuous mode despite this issue, you just need to change the channel with `setCompareChannels(your choice)`, but add a `delay()` before you request the result. The delay needs to be adjusted to the sample rate. The slowest rate is 8 per second. So with `delay(125)` you are on the save side for all sample rates. But I personally don't like this too much.

Potul72, 2020-09-24:
Hi Wollwewald,

Thanks for your fast response.

Let me ask another question. When the SPS are set to for example 120, and
get result is called every second, the register is averaging the previous
120 readings?

Thanks,

Regards,

Roger

wollewald, 2020-09-24:
Hi @Potul72, this would be nice, but is unfortunately not the case. This is a hardware limitation, not a library limitation. This feature would be on my wishlist if someone would design an ADC for me. I would also like to see continuous measurements on all channels in parallel with separate alert thresholds. Haven't found such a device yet. However I still like the ADS1115.

closed 2020-09-30 by wollewald