wollewald/ADS1115_WE #15: ADS1115 - real 16 bit resolution?

hoffmakl1961, 2021-01-25:
On my test module there is the imprint "16 Bit I2C ADC+PGA ADS1115" and the chip on the module has the marking "7A BOGI". 
An evaluation of the measurement results in the conversion register shows that the last 3 bits (D0-D2) are always set to 0.
e.g.
With a gain of PGA 16 (±0.256mV) this corresponds to LSB = 7.8125µV. 
Test results: 
The measuring accuracy is not 7.8125µV, but 125µV. This would be not a resolution of 16 bits. see below my monitoring output. 
With the Adafruit lib: Adafruit_ADS1X15 I don't get a better resolution either.   

**Do you really have a resolution of 16Bit in your test?** 

**My test environment:** 
HW:
ESP32, ADS1115 with 3,3V and MCP4725 with a specially stabilized voltage supply = 3.298V. 

Libs and configuration parameters
I tested at 1st with the Adafruit lib: Adafruit_ADS1X15 and a 2nd test with the lib from Wolfgang Ewald: ADS1115_WE. 
measurement result in each case was the same (the conversion register shows that the last 3 bits (D0-D2) are always set to 0.)

**My tTest environment:** 
HW:
ESP32, ADS1115 with 3,3V and MCP4725 with a specially stabilized voltage supply = 3.298V. 

I create with a MCP4725 e.g. a max. measuring voltage of  256mV, which I can change (reduce) in abaout 60-55µV steps.  
I have connected this measuring voltage with the ADS1115 pin A0 and use the following "configuration parameters":  

adc.setConvRate(ADS1115_8_SPS); 
adc.setMeasureMode(ADS1115_SINGLE);
adc.setCompareChannels(ADS1115_COMP_0_GND);
 
      adc.setVoltageRange_mV(ADS1115_RANGE_0256); // 16x gain  +/- 0.256V  LSBt = 0.0078125mV
      delay(100);
      adc.startSingleMeasurement();  
      while(adc.isBusy()){;}
      volt = adc.getResult_mV();   
      Serial.printf("Channel %u vs GND [V]: %.3fmV\t",Channel , volt);  
      rawValue = adc.getRawResult();      
      Serial.printf("Wandlungsregister: %u\r\n",rawValue);

For each measurement the measuring voltage is reduced by approx. 55µV. 
Here is an excerpt of the monitor output. 

Channel 0 vs GND [V]: 4.625mV	Wandlungsregister: 592
Channel 0 vs GND [V]: 4.625mV	Wandlungsregister: 592
Channel 0 vs GND [V]: 4.625mV	Wandlungsregister: 592
Channel 0 vs GND [V]: 4.500mV	Wandlungsregister: 576
Channel 0 vs GND [V]: 4.500mV	Wandlungsregister: 576
Channel 0 vs GND [V]: 4.500mV	Wandlungsregister: 576
Channel 0 vs GND [V]: 4.375mV	Wandlungsregister: 560
Channel 0 vs GND [V]: 4.375mV	Wandlungsregister: 560
Channel 0 vs GND [V]: 4.250mV	Wandlungsregister: 544
Channel 0 vs GND [V]: 4.250mV	Wandlungsregister: 544
Channel 0 vs GND [V]: 4.125mV	Wandlungsregister: 528
Channel 0 vs GND [V]: 4.125mV	Wandlungsregister: 528
Channel 0 vs GND [V]: 4.125mV	Wandlungsregister: 528
Channel 0 vs GND [V]: 4.000mV	Wandlungsregister: 512
Channel 0 vs GND [V]: 4.000mV	Wandlungsregister: 512
Channel 0 vs GND [V]: 3.875mV	Wandlungsregister: 496
Channel 0 vs GND [V]: 3.875mV	Wandlungsregister: 496
Channel 0 vs GND [V]: 3.750mV	Wandlungsregister: 480
Channel 0 vs GND [V]: 3.750mV	Wandlungsregister: 480

VG Klaus

wollewald, 2021-01-25:
That is weird. **To answer your question first: Yes, I can confirm a resolution of 16 Bits (+/-2^15, to be exact.)**. And I can confirm it also for the +/-256 mV range. Moreover I can confirm the module and the ADS1115 Chip have the same label.

I have used my simple lab power supply device for the voltage to be teested.. I have applied the following parameters:

````
  adc.setVoltageRange_mV(ADS1115_RANGE_0256); 
  adc.setCompareChannels(ADS1115_COMP_0_GND);
  adc.setMeasureMode(ADS1115_CONTINUOUS); 
````

I printed the measured values using:
````
float voltageInMillivolt = adc.getResult_mV(); 
  Serial.print("Result in Millivolt          [mV]: ");
  Serial.println(voltageInMillivolt);

 
  int rawResult = adc.getRawResult();
  Serial.print("Raw Result                       : ");
  Serial.println(rawResult);

  Serial.print("Raw Result, binary:              : ");
  Serial.println(rawResult, BIN);
````

[image]
This shows that the full resolution is used. I repeated it with the single shot mode and like you I used:

````
  adc.startSingleMeasurement();
  while(adc.isBusy()){}
  voltage = adc.getRawResult();
````
This time I printed only the binary values:
[image]

So I have no idea why it's not working properly on your side. 

I have tested another ADS1115 module from different supplier and it also worked. If you want, I can send you a module which works and you send me yours.

hoffmakl1961, 2021-01-26:
Hello Wolfgang,
that's good news that 16 bit resolution is feasible. 

One more question. 
With which ADS1115 operating voltage did you test? 

I have my ADS1115 fixed soldered in. (The goal was that I don't lose µV on the connector).
I still have a 2nd module. But first I have to rebuild everything (soldering).

I will test it first still with it. 

If I get analog results, I will send you the module.

Thanks for your help.

wollewald, 2021-01-26:
I applied 5 volts. IT should work between 2 und 5.5 volts.

Still very strange this behavior. Also the datasheet confirms 16 bit:

[image]

hoffmakl1961, 2021-01-27:
Hello Wolfgang,
for the test I connected the 2nd module (ADS1115) to the PCB and just used a different I2C address. 

I still can't believe it, **in the conversion register the last 3 bits (D0-D2) will be used!!!**

My monitor output: 
Channel 0 vs GND [V]: 0.039mV Conversion register: 5
Channel 0 vs GND [V]: 0.023mV Conversion register: 3

Thanks for your support and your lib.

closed 2021-01-27 by hoffmakl1961

wollewald, 2021-01-27:
Magic! Interesting phenomenon with the first module. Great that it works - have fun!

nikilase, 2021-05-30:
I think I know why he may have had the error on his chip. 
There are many vendors who sell ADS1115 boards that come with an ADS1015 marked as an ADS1115 like mentioned [here](https://forum.arduino.cc/t/achtung-falschungen-ads1115/665025).
My two AliExpress ADS1115 boards also turned out to be fake with an ADS1015 installed. 

Easy test to see if these are fake is seeing the above mentioned behavior of having the raw values always jump by multiples of 16 (because the ADS1015 has 4 bits less resolution).

By the way, even though it may be an ADS1015, it still works fine with the library. 
The only thing that is not working perfect are the Sample Rates. The ADS1015 has higher sample rates going from 128 to 3300 SPS.

wollewald, 2021-05-30:
@nikilase , great thanks that would explain the issue!

wollewald, 2023-01-05:
Hi @nikilase, hi @hoffmakl1961,
the threat us closed quite some time. But I am in discussion (outside GitHub) with someone who also seems to have an ADS1015 instead of the ADS1115. One question: what is the label on the IC of the fake models? In case of an ADS1115 it should be "BOGI". The ADS1015 ICs should be labelled "BRPI" (for VSSOP-types).  I am interested to know if the label of the IC is also fake. 
It would be great if you could check. Thanks!

nikilase, 2023-01-05:
I think I recall I bought the two ADS1115 and also two ADS1118 from this Aliexpress shop https://de.aliexpress.com/item/1005001431623921.html?spm=a2g0o.order_detail.order_detail_item.5.91cd6368SXaKth&gatewayAdapt=glo2deu

It looks like the alleged ADS1115 (rectangle) has the marking "BOGI" and just for your record, the ADS1118 has the marking "BBEI" as you can (hopefully) see in the attached photo.
[image]

I can re-test the chips at a later date when I have time and if you need it :)

kevinjwalters, 2023-07-01:
FYI, for anyone else who finds this issue, there are indeed a lot of fakes out there and I believe many of the sellers know they are selling faulty or misbranded goods. Here's me trying to work out what was going on with bizarre data from a QIFEI branded one: [Adafruit Forums: [Strange Case of Adafruit ADS1115 and QIFEI ADS1115](https://forums.adafruit.com/viewtopic.php?p=853293#p853293)](https://forums.adafruit.com/viewtopic.php?p=869956).