wollewald/ADS1115_WE #16: Connecting another device change the voltage of the other device

chamroeunbm, 2021-03-13:
Hello,
I connect PH sensor to A0 and it works fine(around 1.7v for PH 7)  BUT when I connect EC sensor to A3 the voltage of A0 suddenly drop to 1v. Note when I test them separately i.e connect one by one exclusively, they both work fine. 

Both use 3v input. At first, I suspect that the second(EC Sensor) device consume too much power so I supply voltage to them from another power source but still the same. I also change my ADS1115 chip, but the same.

wollewald, 2021-03-13:
Hi, I have just tried it and all channels work fine. I tried continuous and single shot mode. 

Have you changed the channel using adc.setCompareChannels(ADS1115_COMP_3_GND) before you read the values from channel 3? 

If that's not the issue can you share your code or at least the relevant parts?

chamroeunbm, 2021-03-15:
Sorry, it was due to putting the sensor too near to each other in the same bottle so it interferes each others. Nothing wrong with the ADS.

closed 2021-03-15 by chamroeunbm