wollewald/ADS1115_WE #17: Differential example

luotope, 2021-03-16:
I have managed to succesfully use the library for single ended solutions. Thanks for the great work!
Now I wanted to make a differential test and changed "adc.setCompareChannels(ADS1115_COMP_0_GND)" to "adc.setCompareChannels(ADS1115_COMP_0_1)" and respectively "voltage = readChannel(ADS1115_COMP_0_1)". Then I connect a 3v battery between A0 and A1 but I am getting no reading from it. I couldn't find an example from your github examples either to verify if this was the right thing to do. Could you point me to the right direction and tell where can I find an example for differential measurement with your library. Thanks.

wollewald, 2021-03-16:
Regarding the programming you did the right thing. You just need select the channel and then read the voltage, **BUT** you should never attach a battery or any other voltage supply like this. You need to have a common GND. Don't assume the battery GND (or negative pole to be more exact) is 0 vs the GND of the ADS1115. It can be anywhere. You only know the voltage difference between positive and negative pole of the battery. But you don't know where it vs the module. And since you should never connect a voltage higher than VCC + 0.3 V to the input pins you can easily destroy the module. I hope your module ist still OK. 

This has nothing to do with the library or the module. A common GND is something which is needed fo most circuits. 

If you want to try comparing channels then use e.g. voltage divider between VCC and GND. Or if you want to compare the voltage of two batteries then connect the negative poles of the batteries to the GND of the ADS1115 and the positive poles to the channels you compare.

luotope, 2021-03-16:
Great, thanks. I have now successfully read the (indeed floating) battery voltages between A0 and A1. And thanks to your nice auto scaling function, it shows different batteries voltages very precisely and correctly (compared to, for example, multi-meter reading). I do understand the difference about floating and common ground. What I do not really get yet is: if I should rather use common ground, how would I benefit the (more accurate?) differential measurement?

However, my goal is not to measure batteries (even that was an interesting exercise), but to read small voltages over a 75mV shunt and to create the most precise mA or uA readings for my power supply in for my electronics hobby lab. As I understand it currently, using differential reading instead of single_ended, I can remove some noise etc. Perhaps I need to investigate this a bit more.

Thanks again for your quick response and help. The library is great :)

wollewald, 2021-03-16:
I am not sure how much noise you can reduce to be honest. Interestingly I have done exactly what you intend to do and described it here:

[Self-built Current Sensor](https://wolles-elektronikkiste.de/en/current-sensor-how-to-build-it-yourself)

When I worked on this, I first forgot the common ground of the ADS1115 and the circuit to be measured and wondered why it did not work!

luotope, 2021-03-17:
Thanks for the hint. I can certainly re-use some of your findings from the current sensor project 👍. I do have a 10A 75mA shunt for this which will be on low side. That means that the other end of the shunt will be the same as ground. I now wonder if there is any reason to use differential measurement with A0 and A1 as single end should show the same. However, I have seen some small readings with single ended, even if having the channel grounded, so not sure what will be the lowest level of current which this ADS1115 will ”see”.  But that remains to be seen when I will get that far 😊

wollewald, 2021-03-25:
I think I can close the issue?! Hope your project is successful!

closed 2021-03-25 by wollewald