wollewald/ADS1115_WE #19: different gain settings by channel?

rin67630, 2021-04-05:
Apparently the ADS1115 is not conceived to memorise different gain settings by channel, is it?
A typical application would have been:

- 6V for voltage measurement on channel 1 and 
- 250mV for current measurement over a shunt on channel 2.

How would you proceed to achieve that?
Change the parameters in loop() on the fly and work with one shot conversions?
Could you imagine a solution in the library? 
Thank you for any advice
Regards, 
Laszlo

wollewald, 2021-04-05:
Yes correct, there is no register for this to store the information. So you would have to work with an case/switch or if, like:

if(channel == ADS1115_COMP_0_1){setVoltageRange_mV(range)}

Another option is use setAutoRange(); when you change the channel. 

Or you modify the general function to read the channels in the example sketch:

````
float readChannel(ADS1115_MUX channel) {
  float voltage = 0.0;
  adc.setCompareChannels(channel);
  voltage = adc.getResult_V(); // alternative: getResult_mV for Millivolt
  return voltage;
}
````
**to** individual channels: 

````
float readChannel_1() {
  float voltage = 0.0;
  adc.setCompareChannels(channel_1);
  setVoltageRange_mV(channel_1_range);
  voltage = adc.getResult_V(); // alternative: getResult_mV for Millivolt
  return voltage;
}
````

**or** you extend the function to read the channel:

````
float readChannel(ADS1115_MUX channel, ADS1115_RANGE range) {
  float voltage = 0.0;
  adc.setCompareChannels(channel);
  setVoltageRange_mV(range);
  voltage = adc.getResult_V(); // alternative: getResult_mV for Millivolt
  return voltage;
}
````

and call readChannel accordingly, for example:

````
readChannel(ADS1115_COMP_0_GND, ADS1115_RANGE_2048);
````
If I implement this in the library it will take some additional memory space for the variables. And the effort/complexity on your side does not differ much, since you still have to define somewhere which range you want to apply for which channel. But happy to hear your view. Maybe it would be an alternative to add this as another example sketch?

rin67630, 2021-04-05:
Thank you.
Will it react to a change of setVoltageRange_mV(range); immediately or could a change immediately followed by an adc.getResult() lead to errors?
Could you change it in setAutoRange() without delay?

wollewald, 2021-04-05:
setVoltageRange shouldn't need  a delay. Why don't you just try it? And yes, you can use setAutoRange() instead. It changes into the full range, measures the voltage and then selects the lowest range in which  the measured is below 80% of the maximum (to have some buffer). The only disadvantage is that it takes some conversion cycles. But you don't need  to add a delay.

wollewald, 2021-04-05:
I need to correct myself: setVoltageRange() **needs** a delay. But if you first change the range and then the channel, the delays I have implemented for the change of the channel are sufficient. So do it this way round or take the setAutoRange() function. Sorry for the confusion.

rin67630, 2021-04-06:
Thank you. I wrote here to share our experience and let others benefit as well
I prefer not it use AutoRange to preserve performance (minimize the processing time).
Once I am through, maybe we could add an example?

wollewald, 2021-04-06:
Yes, let's add an example. I'm also considering to add an automatic delay for the change of range.

wollewald, 2021-04-06:
@rin67630 I have modified the library a bit and now you can call setVoltageRange_mV and immediately request voltages. The necessary delays are added in the background. The delays are adjusted to the conversion rate.