wollewald/ADS1115_WE #21: Why autorange blocks so much time when range is under 80% max?

Akrai, 2021-04-20:
So I've noticed that using autorange takes quite a long time. I thought that the function checks if the input is over 80% of the reference voltage before setting the max range, but reading the setautorange function I see that it always changes the reference voltage to the max and then sets it to the optimum, but changing the reference voltage makes use of delay(), so the result is that if I want to do a periodic autorange within the loop(), it blocks way too much time if i decide to use the lower end of the SPS choices.

Wouldn't it be better to not always set the reference to max and then set the optimum, but just try to check before setting a different reference? Or try to avoid delay() in the other function...
If we have the reference voltage value, and the measured value, we can calculate the optimum (if the optimum is a minor value than the current one) without having to change the reference, and if the current value is >80%, then set the maximum and run the function as it is right now

wollewald, 2021-04-21:
I have added a new function: setPermanentAutoRangeMode( true / false). You only have to call it once. See also the Auto_range.ino example. It tries the current range first. Only when the measured value is outside of 30 - 80% of the current range it will call setAutoRange. Please try.

Akrai, 2021-04-22:
It's just what I was looking for! Nice!
I have just found a bug tough: if the measure is over 80% being in the 6144 range, it will keep autoranging forever, when it would avoid autoranging if it is over 80% and being in the 6144 range. It doesn't happen in the 256 range and measuring under 30%

wollewald, 2021-04-22:
@Akrai - Fine, because I just tried it and couldn't find the error. I used one channel, went up and down with the voltage several times and it worked. But please inform me if the bug occurs again. But for now I will close the issue.
And thanks - good recommendation!

closed 2021-04-22 by wollewald