wollewald/ADS1115_WE #25: Using Wire1 with teensy 4.0

ElfyShort, 2022-11-17:
Hi.

First of all I would like to say that I really like this library and I have several projects based on it.
The continues sampling mode is awesome!
Nowadays due to shortage of teensy's I had to adapt some projects to work with teensy 4.0. Unfortunately, the pins (pin 16 and 17) that were previously used I2C wire 0 channel now hooked to wire1 in 4.0.
Is it possible to make this library work based on wire1?
Any help would be appreciated.
Thank you!

ElfyShort, 2022-11-17:
Just had to ask it :)
Fixed it by using
ADS1115_WE adc = ADS1115_WE(&Wire1, I2C_ADDR);

closed 2022-11-17 by ElfyShort

wollewald, 2022-11-17:
Hi @ElfyShort , good to hear you solved it yourself! Thank you for your nice comments on the library.

ElfyShort, 2022-11-17:
Thank you @wollewald for making such a nice library with great examples :)