wollewald/ADS1115_WE #28: Use with ADS1015?

ameliaritger, 2023-12-09:
Just wondering if this library can be used with the 12-bit ADS1015, or if I need to adjust the library for use with my 12-bit board.

wollewald, 2023-12-10:
@ameliaritger ,  the only rlevant differences between the ADS1115 and the ADS1015 is the resolution and the conversion rates. The resolution is no problem for the library. The problem is the conversion rate. If you don't change it then it should be 1600 SPS for the ADS1015 instead of 128 for the ADS1115. What then doesn't work properly is the waiting times that I have implemented after changing channels.

I will make my library fully compatible with the ADS1015 in the next days.

wollewald, 2023-12-10:
@ameliaritger Forgot to add: the different waiting times do not lead to wrong values, since the ADS1015 is faster. You are just losing a bit of time.

ameliaritger, 2023-12-11:
Thanks, I'll have to check my own ADS1015 labels!

wollewald, 2023-12-14:
Short update: it is more difficult than expected to get "real" ADS1015  modules. The one I just received were also ADS1115 based. I am still awaiting Adafruit modules. I expect them to be better than the China stuff. We'll see.

appletv850, 2023-12-17:
> Just wondering if this library can be used with the 12-bit ADS1015, or if I need to adjust the library for use with my 12-bit board.

wollewald, 2023-12-17:
@appletv850 , it should work with ADS1015 with few restrictions. The only difference between the ADS1115 and the ADS1015 is the resolution and the sample rate. The resolution does not matter for the library. But the sample rates are different. And therefore  I will need to adjust the sample rate options. The "translation" is: 

ADS1115_8_SPS          = 128 SPS on the ADS1015              
ADS1115_16_SPS        = 250 SPS on the ADS1015
ADS1115_32_SPS        = 490 SPS on the ADS1015
ADS1115_64_SPS        = 920 SPS on the ADS1015
ADS1115_128_SPS      = 1600 SPS on the ADS1015
ADS1115_250_SPS      = 2400 SPS on the ADS1015
ADS1115_475_SPS      = 3300 SPS on the ADS1015
ADS1115_860_SPS      = 3300 SPS on the ADS1015 (no typo, this is double).

So, if you can live for the moment with this, you should be able to work with the library. 

What holds me back from implementing this is simply that I don't have a "real" ADS1015. I ordered modules from two sources and it turned out they were based on ADS1115 ICs. I have oredered addtional ones from Pimoroni and Adafruit and hope these are better. They should arrive in the next few days and then I launch an update of my library. 

You may also want to test if you really have an ADS1015 with the following sketch and connect A0 to any voltage, but not to GND. 

````
#include<ADS1115_WE.h> 
#include<Wire.h>
#define I2C_ADDRESS 0x48

ADS1115_WE adc = ADS1115_WE(I2C_ADDRESS);

void setup() {
  Wire.begin();
  Serial.begin(9600);
  if(!adc.init()){
    Serial.println("ADS1115 not connected!");
  }
  adc.setVoltageRange_mV(ADS1115_RANGE_6144); 
  adc.setCompareChannels(ADS1115_COMP_0_GND); 
  Serial.println("ADS1115/ADS1015 Example Sketch - Who am I");
  Serial.println("Performing 10 single ended conversions A0 vs. GND:");
  uint16_t checkSum = 0;
  for(int i=0; i<10; i++){
    adc.startSingleMeasurement();
    while(adc.isBusy()){}
    int16_t raw = adc.getRawResult();
    Serial.println(raw, BIN);
    checkSum += raw & 0xF;
  }
  Serial.println();
  Serial.print("Check Sum (Sum of the last 4 bits): ");
  Serial.println(checkSum);

  adc.setConvRate(ADS1115_8_SPS); // = ADS1015_128_SPS
  unsigned long startingTime = millis();
  for(int i=0; i<10; i++){
    adc.startSingleMeasurement();
    while(adc.isBusy()){}
  }
  unsigned long duration = millis() - startingTime;
  Serial.print("Time needed for 10 conversions at slowest sample rate: ");
  Serial.println(duration);
  Serial.println();

  if(checkSum && duration > 1000){
    Serial.println("I am an ADS1115");
  }
  else if (!checkSum && duration < 1000){
    Serial.println("I am an ADS1015");
  }
  else {
    Serial.println("Sorry, don't know who I am!");
  }
}

void loop() {}
````

wollewald, 2023-12-19:
@ameliaritger , @appletv850, finally I received "real" ADS1015 modules. The ADS1015 functionality is implemented in version 1.5.0. You may want to try it.

wollewald, 2024-04-06:
Since there were no further comments I close this issue.

closed 2024-04-06 by wollewald