wollewald/ADS1115_WE #32: Differential result inverted?

realdognose, 2024-04-20:
Hi,

i'm using the ADS1115 to measure two differential channels (0-1 and 2-3) for values upto 0.02 V (a DC-Shunt)
I've quadruple checked the wiring and even used an external meter to verify the result. 

Using the following code (maybe somethings wrong with that?) gives me inverted Results:

```
void loop() {
  commonLoopStart();

  float voltage1 = 0.0;
  float voltage2 = 0.0;
  float relation = 1.0;

  voltage1 = readChannel(ADS1115_COMP_0_1);
  voltage2 = readChannel(ADS1115_COMP_2_3);

  if (voltage2 != 0){
    relation = voltage1/voltage2;
  }

  mqttPublish("pv/voltageChannel1", String(voltage1), true);
  mqttPublish("pv/voltageChannel2", String(voltage2), true);
  mqttPublish("pv/voltageRelation", String(relation), true);

  delay(10000);

  commonLoopEnd();
}

float readChannel(ADS1115_MUX channel) {
  float voltage = 0.0;
  ADS.setCompareChannels(channel);
  delay(100);
  voltage = ADS.getResult_mV(); // alternative: getResult_mV for Millivolt
  return voltage;
}
```

The Multimeter is showing Positive / negative Values between A0_A1 respectively A2_A3 correctly, 
getResult_mV() however returns negative voltages?

[image]

wollewald, 2024-04-20:
@realdognose, I have just tried the following sketch:

````
#include<ADS1115_WE.h> 
#include<Wire.h>
#define I2C_ADDRESS 0x48
ADS1115_WE adc = ADS1115_WE(I2C_ADDRESS);

void setup() {
  Wire.begin();
  Serial.begin(9600);
  if(!adc.init()){
    Serial.println("ADS1115 not connected!");
  }
  adc.setVoltageRange_mV(ADS1115_RANGE_6144); //comment line/change parameter to change range
  adc.setCompareChannels(ADS1115_COMP_0_GND); //comment line/change parameter to change channel
  adc.setMeasureMode(ADS1115_CONTINUOUS); //comment line/change parameter to change mode
}

void loop() {
  float voltage = 0.0;

  Serial.print("0_1: ");
  voltage = readChannel(ADS1115_COMP_0_1);
  Serial.print(voltage);

  Serial.print(",   2_3: ");
  voltage = readChannel(ADS1115_COMP_2_3);
  Serial.println(voltage);
  delay(1000);
}

float readChannel(ADS1115_MUX channel) {
  float voltage = 0.0;
  adc.setCompareChannels(channel);
  voltage = adc.getResult_V(); // alternative: getResult_mV for Millivolt
  return voltage;
}
````
And I have applied this circuit:

[image]

And this is what I get as output:
[image]

Both positive as expected and same result on the multimeter. Repeated with mV Range and still a match (< 1mV deviation). So, all OK on my side. That leads to the difficult question what leads to the negative values on your side and honestly speaking I have no idea. 

With regards to the link you sent: according to my expereince, the data sheet and my latest results A0 and A2 are the positive inputs to get positive results. 

What you could do is checking the wiring on the module, i.e. is the resistance between the Ax (with x = 0-3) Pin of the ADS1115 IC and and the Ax Pin that you connect to your voltage to be measured zero? I have never seen or heard that someone had a wrongly wired module, but not sure what else it could be.

realdognose, 2024-04-22:
hmm, last thing I could check is the actual supply of the AD-Board. Maybe flipping VCC+ and GND makes it produce inverted results and that is what happened? (Not sure, would rather expect it to die immediately, if that would have happened)

Will verify when in range of the device again. 

(For now, adding a *-1 obvisouly worked out as well) 

```
With regards to the link you sent: according to my expereince, the data sheet and my latest results A0 and A2 are the positive inputs to get positive results.
```
Yeah, I also think the datasheet is right and clear on this.

wollewald, 2025-02-08:
@realdognose , since I have not heard any news I close this issue.

closed 2025-02-08 by wollewald