wollewald/ADS1115_WE #36: Sign conversion issue

tronar, 2025-08-24:
I'm very confused because to the best of my knowledge the code is right (library code), yet using this library (1.5.4) 
with an ADS with simmilar input levels in A0 and A3 gives 65520 as output of:

   adc.setCompareChannels(ADS1115_COMP_0_3); 
    adc.startSingleMeasurement();
    while(adc.isBusy()){delay(0);}
    v = adc.getResult_mV();
    SerialMon.println("ADCraw :" + String(v) );

wollewald, 2025-08-24:
Not sure if I understand the issue correctly. Your output is always 65520, independent of the input? I have tried to replicate your code. The complete program is:

````
#include<ADS1115_WE.h> 
#include<Wire.h>
#define I2C_ADDRESS 0x48
ADS1115_WE adc = ADS1115_WE(I2C_ADDRESS);

void setup() {
  Wire.begin();
  Serial.begin(9600);
  adc.init();
  adc.setVoltageRange_mV(ADS1115_RANGE_6144); //comment line/change parameter to change range
  adc.setCompareChannels(ADS1115_COMP_0_3); //uncomment if you want to change the default
  adc.setMeasureMode(ADS1115_SINGLE); //uncomment if you want to change the default
}

void loop() {
  adc.startSingleMeasurement();
  while(adc.isBusy()){delay(0);}
  float v = adc.getResult_mV();
  Serial.println("ADCraw :" + String(v) );

  delay(1000);
}
````
The output I get is:

[image]

adc.getResult_mV() returns a float. I have no idea why you get 65520. This makes no sense. I think you need to provide more details: complete program, which microcontroller / board.

tronar, 2025-08-24:
As I said, I'm confused because the value I got is out of bounds for the configuration.
The setup is a ZMPT101B module feeding A0, A3 from a resistive 2x10k divisor to have differential signed readings.

#define ADS_SCL 26
#define ADS_SDA 27
#define ADS_ALRT 14
#define ADS_I2C 0x48

ADS1115_WE adc = ADS1115_WE(ADS_I2C);

Code in setup is:

  Wire.begin(ADS_SDA, ADS_SCL);
  if(!adc.init()){
    SerialMon.println("ADS1115 not connected!");
    delay(5000);
    ESP.restart();
    delay(5000);
  }
  adc.setVoltageRange_mV(ADS1115_RANGE_2048);

I'm doing a mains voltage meter, used esp-32 ADC with fair results in controlled temp (inside) environment, and then exposed to real world :) So I'm switching to ADS1115.
I'm using Arduino IDE, something is not right but I can not pinpoint what.

wollewald, 2025-08-25:
Still I don't know why you get 65520 as a return value of getResult_mV(), since it returns a float value. 65520 seems to be a negative int16_t. I would expect this to be a return value of getRawResult().

In your code snippets above I can't see how you have defined your variable "v". That's why I asked for the complete program. Maybe your code converts the float into int.

- Can you provide the complete program?
- What do you see if you apply my code?
- Can you try not converting v into a string? I.e. you change:

Serial.println("ADCraw :" + String(v) ); 

into:

Serial.print("ADCraw :"); Serial.println(v);

tronar, 2025-08-25:
I'm so sorry. v was declared as float in the main function, but as unsigned int in a smaller scope, that I missed.
And the behaviour was, well, confusing to me at the time. Thank you for paying attention. That was in the middle of testing the "new" ADC in the old measuring code.

wollewald, 2025-08-25:
No worries! Happy to hear that it is solved.

tronar, 2025-08-25:
Hi, thanks again. I posted in discussions because seemed the better place to abuse your knowledge. Just letting you know here in case there is no notification there ? (and to cover my anxiety, sorry)